mavlink/mavlink #2156: Publish Gimbaled LRF distance through mavlink camera protocol

dirksavage88, 2024-09-30:
Publish Gimbaled LRF distance through mavlink, instead of just the backend mount message

Currently the gimbal LRF distances for the viewpro and SIYI are published via an ardupilot specific mount message, see

[ardupilot/libraries/AP_Mount/AP_Mount_Backend.cpp](https://github.com/ArduPilot/ardupilot/blob/c40422cde32cd3b78630fcf277fa9dce4486e1e3/libraries/AP_Mount/AP_Mount_Backend.cpp#L477)

However this implementation just passively publishes the distance values, whereas these range values can be consumed by some (LightwareLidar is one such gimbal LRF) mavlink-enabled gimbal/camera payloads to focus on a POI and would be a valid use case for these LRF values given the mavlink camera protocol v2.

Instead of only publishing a custom mount message specific to ardupilot, publish via [MAV_CMD_SET_CAMERA_FOCUS](https://mavlink.io/en/messages/common.html#MAV_CMD_SET_CAMERA_FOCUS) with focus type set to range and the value in meters from the POI

modifying the current implementation to publish gimbal LRF values to[ MAV_CMD_TRACKING_GEO_STATUS](https://mavlink.io/en/messages/common.html#CAMERA_TRACKING_GEO_STATUS), however the camera protocol/manager would have to be actively tracking to send distance values for focusing?

The distances reported from the autopilot would be streamed to a compananion/offboard compute that has direct control of camera focusing (it fuses the range from the lidar into a camera parameter to focus at that range)
@hamishwillee

hamishwillee, 2024-10-01:
Hi @dirksavage88 

Let me see if I have this correct (before I try answer).
1. A gimbal might include a lidar range finder (LRF)
2. At the moment these are published by ArduPilot in specific mount messages
3. You would like to publish this information in a standard way so that it can be used by the camera  for things such as setting camera focus and point tracking.

Is that right? I can imagine it is useful for other things as well, such as accurately determining mapping between a captured image and global locations. 

From a terminology point of view, using "publish" above is a bit confusing to me. We generally would use publish or stream to mean telemetry - which is what the distance information would be. We would use "set" or command for things like [MAV_CMD_SET_CAMERA_FOCUS](https://mavlink.io/en/messages/common.html#MAV_CMD_SET_CAMERA_FOCUS).

Could/should you use [DISTANCE_SENSOR](https://mavlink.io/en/messages/common.html#DISTANCE_SENSOR) for this publishing of the LRF info? Or do you really need some direct association with the gimbal for this to work?

FYI @julianoes is the best person to talk camera/gimbal stuff, but let's see if we can clarify the problem statement before he arrives.

julianoes, 2024-10-03:
@dirksavage88 can you give some context what this is used for and how it works, just so I understand what this is for?

auturgy, 2024-10-09:
@peterbarker @rmackay9 I'm not convinced that this is the right approach, however it is relevant information to send through mavlink.  Thoughts?

peterbarker, 2024-10-14:
> Publish Gimbaled LRF distance through mavlink, instead of just the backend mount message
>
> Currently the gimbal LRF distances for the viewpro and SIYI are published via an ardupilot specific mount [...]

That's not really "publishing", at least in a real-time sense.  That's just logging the range.

> However this implementation just passively publishes the distance values, whereas these range values can be consumed by some (LightwareLidar is one such gimbal LRF) mavlink-enabled gimbal/camera [...]

> Instead of only publishing a custom mount message specific to ardupilot, publish via [MAV_CMD_SET_CAMERA_FOCUS](https://mavlink.io/en/messages/common.html#MAV_CMD_SET_CAMERA_FOCUS) with focus type set to range and the value in meters from the POI

Again, ArduPilot's not using that value for anything except logging.

I can't see this going into something like `MAV_CMD_SET_CAMERA_FOCUS`, myself.  Its information, not necessarily a command from the autopilot.

I've often wondered whether ArduPilot should have a RangeFinder backend for these slant-range distance sensors.  So `AP_RangeFinder_MountSlantRange`, which sources range and current attitude from a specified mount instance.  That would then lead to ArduPilot publishing `DISTANCE_SENSOR` messages with a quaternion and distance in it.  A CC might be configured to listen for `DISTANCE_SENSOR` messages with a specific `DISTANCE_SENSOR.id` and choose to adjust focus settings on the camera it is directly controlling.

This is slightly complicated by the fact that ArduPilot can couple mounts and cameras together - for things like the Siyi ZT30, integrated camera+mount setups.  In this case the distance_sensor message could still be consumed by the GCS (MAVProxy's SIYI support could switch to using this message for its slant-range display when we move to using the ArduPilot Siyi support, for example).  However, in these cases we would almost certainly handle the looping back of information (if required?!) internally to the backend driver.  It's really only truly "split-systems", with a mount with a rangefinder in it coupled with an externally-controlled camera where making the slant range information available matters.

> modifying the current implementation to publish gimbal LRF values to[ MAV_CMD_TRACKING_GEO_STATUS](https://mavlink.io/en/messages/common.html#CAMERA_TRACKING_GEO_STATUS), however the camera [...]

... as mentioned above, mavlink already has support for streaming distances.  If a mount has an attached rangefinder - we can publish that already.  This range might be useful for things other than camera focus.

hamishwillee, 2024-10-16:
@peterbarker I think you are saying
- For a separate gimbal/camera we'd expert via [DISTANCE_SENSOR](https://mavlink.io/en/messages/common.html#DISTANCE_SENSOR) - any integration and association would be managed by consumers.
- For integrated camera gimbal we might loopback internally to use the information directly

I tend to think if it is desirable to expert the information in one case then it is desirable in both, because we don't (yet) know how the information will be used - exporting it allows this decision to be deferred.

Anyway, the summary is that you don't this this message is the right approach either - right?

hamishwillee, 2024-10-23:
OK. We're closing this as "probably not the right way to achieve what you want".