mavlink/mavlink #2159: Add a new GPS_FIX_TYPE for RTX satellite corrections

Ryanf55, 2024-10-17:
## Feature request

**Is your feature request related to a problem? Please describe.**

Trimble [PX-1](https://advancedairmobility.trimble.com/) is a GNSS-INS that can receive corrections from satellites. It has the same accuracy as RTK receivers, but is NOT using RTK. ArduPilot does not have an enum to define this kind of receiver. These codes are driven off the MAVlink spec.

**Describe the solution you'd like**

Can we add a new [GPS_FIX_TYPE](https://mavlink.io/en/messages/common.html#GPS_FIX_TYPE) for satellite corrections that allows those kinds of GPS's to show the correct fix type in Mission Planner?

**Describe alternatives you've considered**

Using RTK float  to represent RTX which is misleading.
[image]

**Additional context**

[Centerpoint RTX](https://advancedairmobility.trimble.com/centerpoint.html) is Trimble's marketing name for the GPS type. Perhaps there's a more generic name we can call it? They are currently the only vendor I know that supports this. 

I also checked `gpsd`, but `gpsd` does not support RTX either (yet) even though they support Trimble hardware.

hamishwillee, 2024-10-23:
@Ryanf55 Thanks for this. I think this is what is referred to as [GPS_FIX_TYPE_PPP](https://mavlink.io/en/messages/common.html#GPS_FIX_TYPE_PPP) - this is even mentioned in https://advancedairmobility.trimble.com/centerpoint.html

If this is just about letting the user "know" then that should be fine.

If the GCS needs to do something in particular to help process the data as it does for RTK then we might need a new entry. But would need to know more about the integration.

hamishwillee, 2024-10-23:
Closing - if you think I have this wrong, then please reopen and we can discuss!

closed 2024-10-23 by hamishwillee