mavlink/mavlink #2167: Clarify AMSL Geoid and WGS-84 Ellipsoid Height

Ryanf55, 2024-11-06:
As I understand it mavlink only specifies AMSL for all but a few cases (i.e. wherever you're using a global frame [MAV_FRAME_GLOBAL](https://mavlink.io/en/messages/common.html#MAV_FRAME_GLOBAL). That's a reference independent of the standard ellipsoids. It is supported by most GPS modules, though whether they support it in the same way is unknown (to me).

A few messages such as  [GPS_RAW_INT](https://mavlink.io/en/messages/common.html#GPS_RAW_INT) specify lat/lon in (WGS84, EGM96 ellipsoid) and there are fields for AMSL and alt above above WGS84, EGM96 ellipsoid.

_Originally posted by @hamishwillee in https://github.com/mavlink/mavlink/issues/256#issuecomment-2458653955_

Here are some examples of confusion:

* In [GPS_RAW_INT.lat](https://mavlink.io/en/messages/common.html#GPS_RAW_INT), it says latitude is in `WGS84, EGM96 ellipsoid`. `EGM96` is NOT an ellipsoid, it's a geoid.
* In [GPS_RAW_INT.alt](https://mavlink.io/en/messages/common.html#GPS_RAW_INT), it says `alt` is in `MSL` but does not specify which geoid. I assume EGM-96. This is the only field in the message that needs to clarify which geoid. 
* In [GPS_RAW_INT.alt_ellipsoid](https://mavlink.io/en/messages/common.html#GPS_RAW_INT), it says `Altitude (above WGS84, EGM96 ellipsoid)`. Again, `EGM96` is a goeoid, not an ellipsoid.
* In [GLOBAL_POSITION_INT.at](https://mavlink.io/en/messages/common.html#GLOBAL_POSITION_INT), it says alttidue is in MSL, but does not say which geoid is used. 

ROS 2 has a convention to use WGS-84 ellipsoid height for its geographic coordinates, which is more portable.
https://github.com/ros-geographic-info/geographic_info/blob/ros2/geographic_msgs/msg/GeoPoint.msg

Without specifying which datum the autopilot uses, when it publishes its state estimate in global coordinates, it's impossible to convert to a GeoPoint without making assumptions. 

Not all GPS's use EGM96. Not all terrain data is in EGM96.

What I am hoping to clarify in this ticket is:
* Should we make it more clear that MSL is only supported on the  EGM96 geoid, or should we tell integrators they need to check across their systems they use the same geoid
* Can we remove the mixups of calling EGM96 an ellipsoid?
* Can we come up with a design on how a ROS developer populates a GeoPoint message for the vehicle location given the existing mavlink data from the autopilot?

Ryanf55, 2024-11-06:
Tangentionally related to  https://github.com/PX4/PX4-Autopilot/pull/23854

Ryanf55, 2024-11-06:
MAVROS handles it here:
https://github.com/mavlink/mavros/blob/b49095727a6ff160e1e913b90a4ce50e383e8863/mavros/src/plugins/global_position.cpp#L289C71-L289C96

It hard codes the geoid to EGM-96:
https://github.com/mavlink/mavros/blob/b49095727a6ff160e1e913b90a4ce50e383e8863/mavros/include/mavros/mavros_uas.hpp#L163

hamishwillee, 2024-11-07:
@Ryanf55 Thanks for posting.  I support the goal - we should be precise where possible. Whether we can do it largely depends on whether 
- this will break assumptions that are made by existing users
- the information is actually available to allow conversion to a consistent format.

Currently the assumption is that GPS modules output MSL and this is calculated by some consistent formula.
If MSL is always a proxy for EGM96 we're fine. IF not, then we need to think a bit more about what we are expecting implementers to actually do here, and what implications that would have on drones/SDKs that have been getting the data in some other format.

Let's discuss this in the PR https://github.com/mavlink/mavlink/pull/2168

(note, I'm just pulling people together, I have no idea the answers to this question).