mavlink/mavlink #2171: Mavlink 2.0 C++ 11 error

jiangchunyu, 2024-11-08:
[main] 正在生成文件夹: D:/jcy-woozoom/workspace/DroneServer/out/build/DroneServer DroneServer
[build] 正在启动生成
[proc] 执行命令: "D:\Program Files\cmake-3.30.4-windows-x86_64\bin\cmake.EXE" --build D:/jcy-woozoom/workspace/DroneServer/out/build/DroneServer --parallel 22 --target DroneServer --
[build] [  5%] Building CXX object _deps/fmt-build/CMakeFiles/fmt.dir/src/format.cc.obj
[build] [ 10%] Building CXX object _deps/fmt-build/CMakeFiles/fmt.dir/src/os.cc.obj
[build] [ 15%] Linking CXX static library libfmtd.a
[build] [ 15%] Built target fmt
[build] [ 21%] Building CXX object _deps/spdlog-build/CMakeFiles/spdlog.dir/src/spdlog.cpp.obj
[build] [ 26%] Building CXX object _deps/spdlog-build/CMakeFiles/spdlog.dir/src/stdout_sinks.cpp.obj
[build] [ 31%] Building CXX object _deps/spdlog-build/CMakeFiles/spdlog.dir/src/color_sinks.cpp.obj
[build] [ 36%] Building CXX object _deps/spdlog-build/CMakeFiles/spdlog.dir/src/file_sinks.cpp.obj
[build] [ 42%] Building CXX object _deps/spdlog-build/CMakeFiles/spdlog.dir/src/async.cpp.obj
[build] [ 47%] Building CXX object _deps/spdlog-build/CMakeFiles/spdlog.dir/src/cfg.cpp.obj
[build] [ 52%] Linking CXX static library libspdlogd.a
[build] [ 52%] Built target spdlog
[build] [ 57%] Building CXX object CMakeFiles/DroneServer.dir/src/connect/serial_port/serial_manager.cpp.obj
[build] [ 63%] Building CXX object CMakeFiles/DroneServer.dir/src/connect/tcp/tcp_client.cpp.obj
[build] [ 68%] Building CXX object CMakeFiles/DroneServer.dir/src/connect/tcp/tcp_server.cpp.obj
[build] [ 73%] Building CXX object CMakeFiles/DroneServer.dir/src/connect/udp/udp_client.cpp.obj
[build] [ 78%] Building CXX object CMakeFiles/DroneServer.dir/src/connect/udp/udp_server.cpp.obj
[build] [ 84%] Building CXX object CMakeFiles/DroneServer.dir/src/main.cpp.obj
[build] [ 89%] Building CXX object CMakeFiles/DroneServer.dir/src/modules/uav/uav_module.cpp.obj
[build] [ 94%] Building CXX object CMakeFiles/DroneServer.dir/src/system_manager.cpp.obj
[build] In file included from D:/jcy-woozoom/workspace/DroneServer/include/mavlink/protocol.h:4,
[build]                  from D:/jcy-woozoom/workspace/DroneServer/include/mavlink/common/common.h:29,
[build]                  from D:/jcy-woozoom/workspace/DroneServer/include/mavlink/common/mavlink.h:32,
[build]                  from D:\jcy-woozoom\workspace\DroneServer\src\modules\uav\uav_module.hpp:15,
[build]                  from D:\jcy-woozoom\workspace\DroneServer\src\modules\uav\uav_module.cpp:1:
[build] D:/jcy-woozoom/workspace/DroneServer/include/mavlink/mavlink_types.h:78:12: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
[build]    78 |     struct {
[build]       |            ^
[build] D:/jcy-woozoom/workspace/DroneServer/include/mavlink/mavlink_types.h:17:40: note: in definition of macro 'MAVPACKED'
[build]    17 |   #define MAVPACKED( __Declaration__ ) __Declaration__ __attribute__((packed))
[build]       |                                        ^~~~~~~~~~~~~~~
[build] D:/jcy-woozoom/workspace/DroneServer/include/mavlink/common/common.h:811:25: error: expected unqualified-id before numeric constant
[build]   811 | #define MAVLINK_VERSION 3
[build]       |                         ^
[build] D:/jcy-woozoom/workspace/DroneServer/include/mavlink/common/common.hpp:27:16: note: in expansion of macro 'MAVLINK_VERSION'
[build]    27 | constexpr auto MAVLINK_VERSION = 3;
[build]       |                ^~~~~~~~~~~~~~~
[build] In file included from D:/Program Files/x86_64-14.2.0-release-posix-seh-ucrt-rt_v12-rev0/mingw64/x86_64-w64-mingw32/include/windows.h:71,
[build]                  from D:/Program Files/x86_64-14.2.0-release-posix-seh-ucrt-rt_v12-rev0/mingw64/x86_64-w64-mingw32/include/winsock2.h:23,
[build]                  from D:/local/boost_1_83_0/prefix/include/boost-1_83/boost/asio/detail/socket_types.hpp:33,
[build]                  from D:/local/boost_1_83_0/prefix/include/boost-1_83/boost/asio/detail/win_event.hpp:25,
[build]                  from D:/local/boost_1_83_0/prefix/include/boost-1_83/boost/asio/detail/event.hpp:23,
[build]                  from D:/local/boost_1_83_0/prefix/include/boost-1_83/boost/asio/execution/blocking_adaptation.hpp:19,
[build]                  from D:/local/boost_1_83_0/prefix/include/boost-1_83/boost/asio/execution.hpp:22,
[build]                  from D:/local/boost_1_83_0/prefix/include/boost-1_83/boost/asio/any_completion_executor.hpp:22,
[build]                  from D:/local/boost_1_83_0/prefix/include/boost-1_83/boost/asio.hpp:20,
[build]                  from D:/jcy-woozoom/workspace/DroneServer/src/connect/serial_port/serial_manager.hpp:12,
[build]                  from D:\jcy-woozoom\workspace\DroneServer\src\modules\uav\uav_module.hpp:10:
[build] D:/jcy-woozoom/workspace/DroneServer/include/mavlink/common/common.hpp:492:5: error: expected identifier before numeric constant
[build]   492 |     ERROR=1, /* generic error / not accepting mission commands at all right now | */
[build]       |     ^~~~~
[build] D:/jcy-woozoom/workspace/DroneServer/include/mavlink/common/common.hpp:492:5: error: expected '}' before numeric constant
[build] In file included from D:\jcy-woozoom\workspace\DroneServer\src\modules\uav\uav_module.hpp:16:
[build] D:/jcy-woozoom/workspace/DroneServer/include/mavlink/common/common.hpp:490:1: note: to match this '{'
[build]   490 | {
[build]       | ^
[build] D:/jcy-woozoom/workspace/DroneServer/include/mavlink/common/common.hpp:492:5: error: expected unqualified-id before numeric constant
[build]   492 |     ERROR=1, /* generic error / not accepting mission commands at all right now | */
[build]       |     ^~~~~
[build] D:/jcy-woozoom/workspace/DroneServer/include/mavlink/common/common.hpp:523:5: error: expected identifier before numeric constant
[build]   523 |     ERROR=3, /* Indicates an error in secondary/redundant systems. | */
[build]       |     ^~~~~
[build] D:/jcy-woozoom/workspace/DroneServer/include/mavlink/common/common.hpp:523:5: error: expected '}' before numeric constant
[build] D:/jcy-woozoom/workspace/DroneServer/include/mavlink/common/common.hpp:519:1: note: to match this '{'
[build]   519 | {
[build]       | ^
[build] D:/jcy-woozoom/workspace/DroneServer/include/mavlink/common/common.hpp:523:5: error: expected unqualified-id before numeric constant
[... 627 lines cut ...]
[build] D:/jcy-woozoom/workspace/DroneServer/include/mavlink/common/common.h:795:14: note: previous definition here
[build]   795 | typedef enum ADSB_FLAGS
[build]       |              ^~~~~~~~~~
[build] D:/jcy-woozoom/workspace/DroneServer/include/mavlink/common/common.hpp:766:16: error: 'constexpr const auto ADSB_FLAGS_ENUM_END' redeclared as different kind of entity
[build]   766 | constexpr auto ADSB_FLAGS_ENUM_END = 65;
[build]       |                ^~~~~~~~~~~~~~~~~~~
[build] D:/jcy-woozoom/workspace/DroneServer/include/mavlink/common/common.h:804:4: note: previous declaration 'ADSB_FLAGS ADSB_FLAGS_ENUM_END'
[build]   804 |    ADSB_FLAGS_ENUM_END=65, /*  | */
[build]       |    ^~~~~~~~~~~~~~~~~~~
[build] D:/jcy-woozoom/workspace/DroneServer/include/mavlink/common/common.hpp:769:1: error: expected declaration before '}' token
[build]   769 | } // namespace common
[build]       | ^
[build] D:/jcy-woozoom/workspace/DroneServer/include/mavlink/common/common.hpp:770:1: error: expected declaration before '}' token
[build]   770 | } // namespace mavlink
[build]       | ^
[build] mingw32-make[3]: *** [CMakeFiles\DroneServer.dir\build.make:166: CMakeFiles/DroneServer.dir/src/modules/uav/uav_module.cpp.obj] Error 1
[build] mingw32-make[3]: *** Waiting for unfinished jobs....
[build] mingw32-make[3]: *** [CMakeFiles\DroneServer.dir\build.make:151: CMakeFiles/DroneServer.dir/src/main.cpp.obj] Error 1
[build] mingw32-make[3]: *** [CMakeFiles\DroneServer.dir\build.make:181: CMakeFiles/DroneServer.dir/src/system_manager.cpp.obj] Error 1
[build] mingw32-make[2]: *** [CMakeFiles\Makefile2:118: CMakeFiles/DroneServer.dir/all] Error 2
[build] mingw32-make[1]: *** [CMakeFiles\Makefile2:125: CMakeFiles/DroneServer.dir/rule] Error 2
[build] mingw32-make: *** [Makefile:168: DroneServer] Error 2
[proc] 命令“"D:\Program Files\cmake-3.30.4-windows-x86_64\bin\cmake.EXE" --build D:/jcy-woozoom/workspace/DroneServer/out/build/DroneServer --parallel 22 --target DroneServer --”已退出，代码为 2
[driver] 生成完毕: 00:00:31.294
[build] 生成已完成，退出代码为 2

hamishwillee, 2024-11-14:
This sounds like a support request that doesn't give enough information to debug. @julianoes Can you advise anything on this? Otherwise please close.

julianoes, 2024-11-14:
You write C++11 error. MAVSDK requires C++17. Maybe that's the problem?

vooon, 2024-11-14:
https://github.com/mavlink/mavros/blob/ros2/libmavconn/include/mavconn/mavlink_dialect.hpp.em

You should **not** include both C and C++11 messages at the same time. Choose one: `.h` or `.hpp`.

vooon, 2024-11-14:
@julianoes no, for generated lib 2011 is enough. https://github.com/ArduPilot/pymavlink/blob/533161c47995ba47f5548636eed37a47e91072c0/generator/CPP11/test/posix/Makefile

julianoes, 2025-09-03:
OP never came back.