mavlink/mavlink #2186: ZOOM_TYPE_RATE

nrd8, 2024-11-30:
I noticed there is currently no option to zoom at different speeds/rates while using MAV_CMD_SET_CAMERA_ZOOM with joysticks. Current options include step, in/out (Continuous) and absolute zoom level (Range, focal length or FOV). My proposal is twofold.

Option A:
Adding a new CAMERA_ZOOM_TYPE -> ZOOM_TYPE_RATE that would function similar to GIMBAL_MANAGER_SET_MANUAL_CONTROL pitch and yaw fields by taking a normalized range [-1..1] representing a zoom rate in this case.

Option B:
Changing the definition of the already existing CAMERA_ZOOM_TYPE -> ZOOM_TYPE_CONTINUOUS to take a normalized range [-1..0] instead of -1 | 0 | 1. This would make ZOOM_TYPE_CONTINUOUS backwards compatible while still adding functionality.

Thank you,
nrd8

hamishwillee, 2024-12-04:
Option B would be breaking because a system that didn't know about it would not understand the intermediate values. 

Option A is possible, as would be adding a field for the rate. 

Why do you want to control the rate - what's the use case? What camera do you have that allows the rate of zoom to be controlled?  Are you working with a camera manager or flight stack vendor that will support this change?

The reason I ask is that there is no point making a change like this without some buy-in from the stakeholders to get an implementation.

hamishwillee, 2024-12-04:
Discussed in the call - we would probably consider option B, but with the range [-1..1], not [-1..0] as written above.

nrd8, 2024-12-04:
Hi hamishwillee,

Thank you for your response. I totally agree that having a range [-1..1] makes more sense. I personally also prefer option B because it naturally fits in with ZOOM_TYPE_CONTINUOUS.

As for the use case. This is useful for gimbals/camera systems running a companion computer that controls the zoom internally. For my case I have a camera with analog zoom capabilities using a motorized lens. Such camera systems may be controlled using a joystick type input enabling the user to zoom in/out fast or make smaller slower adjustments.

hamishwillee, 2024-12-04:
@nrd8 Thanks for the clarification. What camera? What GCS? What flight stack?

The reason I ask is that we have to validate changes before they merge to show that they are fit for purpose and don't break anything. Something on a private branch that only you are using makes that harder. It is also good to be able to point to that stuff when someone says how do you know it works.

Option A is safer IMO, and we might need to do that. If we do option B it _should_ be safe as per the definition, but it might not be as per the implementation. Specifically, it is not uncommon for implementers to treat the discrete values `[-1, 0. 1]` as ranges `[-1-0.5, -0.5 to +0.5, +0.5 to 1]` - I need to convince myself that sending anything in the range will still work as expected. 

My thinking is that it will be OK. If a system hasn't been updated and gets a signal that isn't one of the allowed values it should ignore it. But it might activate if it gets into the next range. That means that if you plugged in a different camera with the old-style zoom you would get nothing, then at about 0.5 it would go full speed, even though you only specified 0.5 (say). I think that is OK?

nrd8, 2024-12-20:
Hi Hamish,
Sorry for the late reply.

Here is some information about my setup. I am using the Optroxa GMB 600 gimbal and SRoC GCS running QGC. No flight stack at the moment.

This is a valid concern IMO and option A will be the safe choice but it might introduce a potentially redundant field. 

Say we go with option B:
If an integrator is using the original style ZOOM_TYPE_CONTINUOUS (discrete) and they receive anything but 0 meaning the controller is trying to change zoom it will resolve to the discrete value (-1,0,1 depending on implementation) resulting in correct behavior. The question in this case is where the threshold for switching from 0 to -1/1 is, if that is an absolute value above 0 or an absolute value above 0.5. Either way it will function as a discrete button as expected when using a joystick. In c++ an absolute value between [0-1[ (1 not included) will resolve to 0 when cast to a signed integer. 

If an integrator is using a "range" interpretation of ZOOM_TYPE_CONTINUOUS (including variations of 0.5) they are already expecting and accounting for floating point values resulting in correct behavior.

These are my thoughts on the matter but I am personally ok with both options as you see most fit for the protocol.

Looking forward to hearing your opinion.

- nrd8

hamishwillee, 2025-01-01:
@nrd8 Thanks! I'm just dropping in for a few days between holiday breaks, so sorry for the delay.

I think you're right - option B seems safe.

Just thinking about this a bit more, the other concern is UX. As things are now, the UI is effectively a three way switch. With rate control it is a slider. So users may get unexpected behaviour - i.e. they set the slider to 0.25 and nothing happens, or they set to 0.75 and it goes max as quickly as it can. Perhaps this could be solved by assuming the current mechanism and allowing the user to choose to make it a slider if they know the behaviour is supported by the camera (The "right" way would be to tell the GCS what kinds of zooms are supported, but the zoom type is an enum not a bitmask, so this is not easy to expose without duplication of the enum)

Are you also planning on updating generic QGC to control the zoom using the new version of the method? Any thoughts on this?

Either way would you like to create a PR so we can present this to the stakeholders for discussion later in the month? If not I can do that when I get back properly from holiday.

nrd8, 2025-01-09:
Hi Hamish,

Hope you had a great holiday!

These are valid points but since option B would only be a definition change nothing would actually change implementation wise for integrators expecting the old style discrete method: -1 | 0 | 1. This is possible because commands already encode the values as floats.

On the other hand if an integrator were to use the new style continous/rate method they are expecting a value between [-1.0 .. 1.0] and would naturally account for these types of inputs.

The problem only occurs if you have a GCS and a platform that don't agree in which case the consequence would be minimal and  you could argue that the integrator isn't following the protocol.

I think it would make sense to get this into generic/vanilla QGC at some point. This would be when QGC decides to update the Mavlink version they are using. For now I'm mostly interested in expanding the Mavlink protocol with this new ZOOM_TYPE capability.

Looking forward to hearing from you.

hamishwillee, 2025-01-23:
Yes integrators are fine - they always get data they know what to do with. If the GCS doesn't support a UI to allow the continuous zoom you won't have any impact other than continuous zoom not working. If the GCS does support UI to allow continuous zoom and the platform doesn't the behaviour is a bit rubbish for users - but that can be solved independent of this change if desired.

So yes, you can create a PR for this change. Note however that we will need to validate/verify that it does work as expected before we can merge. Have you a prototype and thoughts for validation?

nrd8, 2025-04-16:
Hi again,

Sorry for my absence and delay. I have now created a PR on the mavlink-devguide repository with the changes discussed in this thread. I also added the same change for ZOOM_TYPE_FOCUS for consistency.

Link to PR: https://github.com/mavlink/mavlink-devguide/pull/598

Let me know what you think. Thanks.

hamishwillee, 2025-04-30:
Thanks - I've been away too. Note, updated in https://github.com/mavlink/mavlink/pull/2260 - now let's see if we can get it merged

julianoes, 2025-09-03:
PR merged, closing.

closed 2025-09-03 by julianoes