mavlink/mavlink #2199: MAVLink Gimbal v2 protocol

qsivey, 2024-12-24:
Hello. 
I work at BaseCam Electronics (Alexmos), and we’re currently planning to implement the Gimbal v2 protocol into our controllers.
Is there anyone here who could help with this, provide more information, or offer direct assistance?
Thanks in advance!

hamishwillee, 2024-12-27:
The gimbal protocol is quite well documented here: https://mavlink.io/en/services/gimbal_v2.html

If you need free support then the discuss forums here are are a good place to start https://discuss.px4.io/c/mavlink/23

If you need more specific help, the best person I could point you to is @julianoes - he's a contractor, and also one of the authors of this interface. 

Note, likely to be slow response here for a while. We're on holiday.

julianoes, 2024-12-27:
@qsivey it would be great if the gimbal implemented the GIMBAL_DEVICE part of the protocol specifically.

There is also a tester for gimbal devices based on MAVSDK:
https://github.com/mavlink/MAVSDK/tree/main/examples/gimbal_device_tester

julianoes, 2024-12-27:
@qsivey send me an email, happy to help with this.

qsivey, 2024-12-28:
Thank you so much for response, it means a lot to me.
@julianoes I suggest we revisit this issue after the holidays around early January. I also need to finish some other work tasks in the meantime.
My work email address is:
ai@basecamelectronics.com

hamishwillee, 2025-02-06:
@julianoes @qsivey Closing this. If you reconnect and there is any work to be done feel free to reopen

closed 2025-02-06 by hamishwillee