mavlink/mavlink #2201: DO_LAND_START contains lat/lon/alt coordinate which is difficult to maintain

DonLakeFlyer, 2025-01-19:
Mavlink spec says that "If specified then it will be used to help find the closest landing sequence.". This is to support multiple landing sequences and finding the closest landing sequence to the current vehicle position. 

As things currently stand ArduPilot supports multiple landing sequences but does not use the coordinate from DO_LAND_START. It uses the first coordinate it finds after the DO_LAND_START. At this point I don't believe PX4 supports multiple landing sequences in a mission.

The problem relates to how a GCS creates a mission with a landing sequence prefixed with DO_LAND_START. You can see a discussion here about it as well: https://github.com/mavlink/qgroundcontrol/pull/12313. Since the vehicle does not actually fly through the coordinate in DO_LAND_START it means a GCS needs to fill in this coordinate during mission creation. The way it's going to do that is to use the coordinate of the next item. If the user moves that item, then a GCS needs to go back and update to DO_LAND_START with a new value. All of that just leads to major complexity in mission creation.

It would be way easier if instead the spec was changed. Two possibilities:
* Coordinate in DO_LAND_START is optional. If not specified the starting position of the landing sequence is determined from the first coordinate after DO_LAND_START.
* DO_LAND_START has no coordinate. The starting position of the landing sequence is determined from the first coordinate after DO_LAND_START. This is how ArduPilot works.

I would personally go for the second option because it is the simplest. Although it diverges further from the existing spec, but it's closer to the reality of what's out there.

DonLakeFlyer, 2025-01-19:
Would be good if someone from ArduPilot side can verify how it actually works in relation to DO_LAND_START.

auturgy, 2025-01-19:
@tridge @magicrub

hamishwillee, 2025-01-22:
The way I read [MAV_CMD_DO_LAND_START](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_LAND_START) the meaning of lat/lon/altitude for this command as a mission item is undefined. This is ambiguous though - it should make that clear if it is the case

The only meaning is that when sent as a command, where it allows the selection of one among many landing sequences.

As you say though, it depends on what both PX4 and ArduPilot do with that co-ordinate in a mission - if it is ignored in both then I'd suggest we note that they are only used in commands and the first point in the landing sequence is the one in the waypoint that follows this marker.

Otherwise your suggestion would be good.

What does PX4 do in a mission with this information?

peterbarker, 2025-01-22:
The location in `DO_LAND_START` is used to determine *which* `DO_LAND_START` to actually use.  AFAIK the only implementation of the command chooses the *closest* `DO_LAND_START` to jump to.

The coordinates are *not* navigated to, simply used to determine landing sequence which to use.

DonLakeFlyer, 2025-01-22:
@peterbarker Ok, so i was wrong then. You are talking about ArduPilot current implementation right? That would indicate it's too late to change and GCS will just have to deal with this.

@rubenp02 FYI

peterbarker, 2025-01-23:
> [@peterbarker](https://github.com/peterbarker) Ok, so i was wrong then. You are talking about ArduPilot current implementation right? That would indicate it's too late to change and GCS will just have to deal with this.

Yes, sorry, *very* long-standing behaviour in ArduPlane!  We used it in the OBC in 2014, I think!

FWIW, even MAVProxy doesn't deal with DO_LAND_START very well in terms of visual on the map.  We still indicate we're going to try to fly there.  I did have a look at changing the rendering code, but it's non-trivial.

hamishwillee, 2025-01-23:
@peterbarker @DonLakeFlyer Can you confirm that #2203 reflects how this is in ArduPilot/PX4?

@peterbarker Just FYI software that parses the XML can tell if it should render the point based on the isDestination attribute.

[image]

peterbarker, 2025-01-23:
> [@peterbarker](https://github.com/peterbarker) [@DonLakeFlyer](https://github.com/DonLakeFlyer) Can you confirm that [#2203](https://github.com/mavlink/mavlink/pull/2203) reflects how this is in [...]

Yes indeed.  What I was *trying* to do, however, was to show multiple entry-points into the one landing sequence, so you could still see where the DLS existed on the map and could reposition as required.

peterbarker, 2025-01-23:
> However, when I tested this in SITL, I found that even if the coords are set to 0 AP still finds the closest sequence to navigate to. I still kept the code in because of that one line in the MAVLink spec in case some firmware does rely on it, but it'd be helpful to standardise the behaviour as a mission item vs. as a command. From the GCS perspective it'd be much cleaner if coords in the mission item weren't valid, and it makes more sense too.

We corrected that in ArduPlane's 4.2 release (in 2021); prior to that not having a location in `DO_LAND_START` was a Bad Thing.

It's important to allow the user to specify a location in the DLS item; the flexibility may allow for selection of an appropriate path without needing to place the point in such a way that the vehicle can move there safely (ie. avoiding obstacles while taking into account turning radius)

Supporting multiple landing sequences in QGC sounds like a good thing!  I've met ArduPilot users who permanently have multiple landing sequences planned, the vehicle choosing between them automatically depending on wind direction.

It probably won't affect you, but I'll mention this anyway.  There's an additional mission item which ArduPilot supports, `MAV_CMD_DO_RETURN_PATH_START`.  This essentially makes all subsequent waypoints valid places to start returning home.  This is useful for things like pipeline missions.  Demo is here: 1cdff47246b2d60c0301eb36da51946bedc09c18 .  Note that while the message is in common.xml I doubt any flight stack apart from ArduPilot supports it.

DonLakeFlyer, 2025-01-30:
Sorry for the delay, just got back in town.

> and current APM (better to not have them unless the user wants a special behaviour).

I don't quite understand this comment. 

That said I'd propose the following:
* To start you may want to familiarize yourself with this: https://docs.qgroundcontrol.com/master/en/qgc-dev-guide/plan/mission_command_tree.html#mission-command-ui-info. This is what drives the QGC ui for items used to plan a mission.
* Since according to mavlink spec specifying a coord in do land start should be supported, then the default spec for  DO_LAND_START in MavCmdInfoCommon.json should have `specifiesCoordinate: true` and `standaloneCoordinate: true`. This indicates that by default a DO_LAND_START item in a mission has lat/lon/alt values but the vehicle does not fly through it. Similar to an ROI item for example.
* The PX4-MavCmdInfoCommon.json should override both `specifiesCoordinate` and `standaloneCoordinate` to false since PX4 doesn't support a coord.
* In the code which creates the landing complex item mission items, it should interrogate the MissionCommandTree associated with the firmware type of the vehicle the plan is being created for. The entry for `DO_LAND_START` will indicate whether the coord should be filled in or not based on the `specifiesCoordinate` setting.
* I don't think DO_LAND_START's should be automatically futzed with when existing plans are loaded. If old plans are incorrect given the strangeness/differences in firmware they will need to be recreated/resaved to get them fixed up.

Does that make sense?

hamishwillee, 2025-02-06:
@DonLakeFlyer @peterbarker  @rubenp02 

Just to be clear, my understanding of this is that when [MAV_CMD_DO_LAND_START](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_LAND_START) is used _in a mission_ then lat/lon/alt should be ignored, and the actual **start point** of the landing sequence is the lat/lon/alt of the first positional waypoint _after_ the land_start.

That's correct right? (for APM4.1+ and PX4). 

[for AMP before 4.1 I understand the **start point* is lat/lon/alt in the land start message).

When used in a command expected behaviour is unambiguous:
- Triggers a landing
- If lat, lon, alt set to 0 then they are ignored and FC chooses landing sequence with **start point** geographically closest to vehicle.
- if lat, lon, alt set to values, then flight stack chooses landing sequence with **start point** geographically closest to that point.

APM ignores lat/lon/alt currently - so behaves as if set to zero.

Is that all correct? If so I can update #2201 to remove the ambiguity.