mavlink/mavlink #2202: Gimbal v2 protocol: mix yaw from vehicle and pitch from ground station

MaEtUgR, 2025-01-22:
Initial situation: A 3-axis gimbal without limited yaw freedom (landing gear in camera view and physical limit) should be manually piloted. The gimbal is yawed using the vehicle yaw stick and pitched using a dedicated pitch rocker. It's currently done with Gimbal v2 protocol by the ground station sending `GIMBAL_MANAGER_SET_MANUAL_CONTROL` messages with flags set to "lock" roll and pitch in world frame but yaw "unlocked" in body frame which makes the gimbal follow vehicle body frame yaw. Vehicle stick input including yaw stick comes through SBUS and the pitch rate for the gimbal comes from the `GIMBAL_MANAGER_SET_MANUAL_CONTROL` message directly from the ground station.

@Perrrewi and I are working on a feature where you get more accurate control over gimbal yaw. For this purpose in a particular flight mode the gimbal gets commanded to "lock" yaw to the world frame and yaw rate from the vehicle's yaw stick. The vehicle then makes sure to never interfere with the view and follow any yaw movement of the gimbal such that the gimbal can accurately point somewhere ignoring the vehicle. We implemented everything in PX4 and it works nicely but there is one thing not clearly defined by the protocol that we'd like to do right and have supported in MAVLink if possible.

https://github.com/PX4/PX4-Autopilot/pull/24242

The problem: Previously the gimbal control as far as the protocol is concerned was with the ground station, now it's with the autopilot which controls the yaw with the yaw stick. But the pitch input should still come from the ground station without taking away control from the vehicle so it's basically a mixed use.

Options:
1. Not cover this in the gimbal protocol and send over the pitch rocker position separately and assemble it in the flight mode.
2. Allow to send a `GIMBAL_MANAGER_SET_MANUAL_CONTROL` with only pitch rate input even when the ground station is not in control and allow the mode to pick it up and mix the inputs.
3. Add the ground station as secondary control and mix in the pitch rocker that way.

Tagging @julianoes since I'm wondering what his intentions would be.

MaEtUgR, 2025-01-23:
@julianoes Thanks a lot for your input!

> Or leave the ground station as primary and make PX4 add itself as the secondary?

I considered that, where it made me hesitate is PX4 has to set the flags so the flags from the primary control entity would not be considered which seems strange and there's no defined way to not impose any flags like nan for a rate.

> And then mix it in the gimbal driver?

Yeah gimbal manager/module, I also considered that and hesitated with how the mapping between raw input and actual rate would then get scattered. E.g. In that mode the yaw rate is taken from the stick and scaled + filtered such that the sensitivity feels good for the lens and zoom level currently in use. For pitch the scaling and filtering should probably be similar which is easy to do when it happens in the same component but gets hard if the gimbal manager does the same job for pitch... 🙈

I can't eliminate the feeling that one place should be responsible for coordinating/mixing the gimbal and that place needs to be situation-aware, as in what the current user goal is.

Let me think about it more, thanks again 🙏

MaEtUgR, 2025-01-27:
To be honest we needed to have a prototype working quickly for initial testing. And the way we did it is misusing the protocol to have the ground station always send `GIMBAL_MANAGER_SET_MANUAL_CONTROL` messages even when it's not in control, the gimbal manager by definition ignores these messages but we salvage the pitch rate setpoint (pitch knob position) from it in the place we control vehicle and gimbal for that feature:
https://github.com/PX4/PX4-Autopilot/pull/24242/commits/0d51c837353f3df0df8a925226658e6c455f4efe#diff-66efd3d6480dc0421543176a3e2b6b6341c18e6906b11e4173b043ef5b0caf98R137

Now we're thinking through how to solve this properly but it seems overly complicated and I'm not sure if the result wouldn't just limit the next use case that's again different than ours.

The thoughts so far are that it would make sense to have the mode implementation have the primary control and the ground station chime in as secondary control. But it would need to be clear if a secondary control is allowed and everything would need to be mixed correctly again. Here are some high level steps:
1. Ground station sending secondary control only if secondary control desired (RC_MIXED flag?).
2. Gimbal manager mixing control.
  - Gimbal manager needs to know maximum rates (based on settings/focal length) for which there is currently no specific message.
  - Needs to accept pitch only (yaw nan) which is currently not allowed.
  - Needs to use the flags of the primary control.
3. Gimbal manager or gimbal needs to do the input shaping 
4. Vehicle needs to yaw based on gimbal telemetry.
  - Latency ok?
  - Angular rate telemetry could be noisy depending on measurements.

This feels like over-engineering if really we only want to get the pitch position of the RC rocker in the mode implementation for our use case 😬

julianoes, 2025-01-27:
Still trying to wrap my head around this all. Thanks for pushing this along.

Some somewhat random thoughts:

A) Wasn't there some option at some point that did something like that, basically use vehicle yaw instead of gimbal yaw for yaw? I faintly remember a setting. Just so we don't duplicate it but replace it while doing this.

B) You say that the ground station knows the limits based on field of view but it's complicated to send everything to the gimbal driver to do the mixing appropriately, right? Wouldn't the right approach be to do the calculation on the ground side instead and use the [GIMBAL_MANAGER_SET_ATTITUDE](https://mavlink.io/en/messages/common.html#GIMBAL_MANAGER_SET_ATTITUDE) instead?

And trying to answer your points above:

> 2. Needs to accept pitch only (yaw nan) [which is currently not allowed](https://github.com/PX4/PX4-Autopilot/blob/5bca71791a27db30e89e8904c40cab51e11f8e8b/src/modules/gimbal/input_mavlink.cpp#L940). In the protocol nothing speaks against this.

I think it would be fine to do that.

> 4. Vehicle needs to yaw based on gimbal telemetry. 

That doesn't sound quite right indeed. 🤔