mavlink/mavlink #2239: Issues with Receiving GIMBAL_DEVICE_ATTITUDE_STATUS Message on MAVLink

FabianEP11, 2025-03-17:
Hi 
I am currently working on a project using a PH6 and a SIYI ZT6 camera, and I am trying to get values from the gimbal using the GIMBAL_DEVICE_ATTITUDE_STATUS message. However, I am not receiving the expected message consistently. I have the following setup:

`pip show pymavlink`
```
Name: pymavlink
Version: 2.4.42
Summary: Python MAVLink code
Home-page: https://github.com/ArduPilot/pymavlink/
Author: 
Author-email: 
License: LGPLv3
Location: /home/user/miniconda3/envs/project3.10/lib/python3.10/site-packages
Requires: future, lxml
Required-by: MAVProxy
```

A short version of my code: 
```

from pymavlink import mavutil
master = mavutil.mavlink_connection(args.serialport, baud=115200)
# I tried different 
# master = mavutil.mavlink_connection(args.serialport, baud=115200,  dialect='common')         
# master = mavutil.mavlink_connection(args.serialport, baud=115200,  dialect='common', version=2)

master.wait_heartbeat()

# Request GIMBAL_DEVICE_INFORMATION
    time.sleep(1)
    master.mav.command_long_send(
        master.target_system,
        master.target_component,
        mavutil.mavlink.MAV_CMD_REQUEST_MESSAGE,
        0, # confirmation
        285, 0, 0, 0, 0, 0, 0)
# I tried with 283, 285 

while True:
        mavlink_msg = master.recv_match(blocking=True)
        print(f"Received message type: {mavlink_msg.get_type()}")
```

Issue:

By default, the controller is not sending the gimbal information. According to the documentation, I need to send a request, which I’ve done with the MAV_CMD_REQUEST_MESSAGE command. However, when I run the code, I do not receive any GIMBAL_DEVICE_ATTITUDE_STATUS message immediately. Instead, I see  "Received message type: UNKNOWN_285"

[image]

When I kill the node and run it again multiple times, eventually I start receiving gimbal information on the GIMBAL_DEVICE_ATTITUDE_STATUS message

Is there something else I need to do to consistently receive the GIMBAL_DEVICE_ATTITUDE_STATUS message? 
Any guidance or recommendations would be greatly appreciated!
Thank you

hamishwillee, 2025-03-20:
@FabianEP11 What flight stack are you talking to, and in what version? Also simulator or real hardware?

The id of [GIMBAL_DEVICE_INFORMATION](https://mavlink.io/en/messages/common.html#GIMBAL_DEVICE_INFORMATION) is 283.

Requesting 285 will get you [GIMBAL_DEVICE_ATTITUDE_STATUS](https://mavlink.io/en/messages/common.html#GIMBAL_DEVICE_ATTITUDE_STATUS). This should be streamed by your gimbal all the time I believe, so you shouldn't have to request it.
The message `UNKNOWN_285` tells you that this message is not known by Pymavlink, which implies to me that you're talking to ArduPilot. Pymavlink is built using ardupilot's set of messages by default - they probably don't do anything with `GIMBAL_DEVICE_ATTITUDE_STATUS` so they haven't copied it into their common. It is possible to build Pymavlink with the proper definitions - see the docs.

I would be looking for the COMMAND_ACK back from your request first - that should tell you that the request for GIMBAL_DEVICE_INFORMATION was received, and whether the vehicle was able to respond.

@julianoes There isn't anything special you need to do in order to request GIMBAL_DEVICE_INFORMATION right?

hamishwillee, 2025-03-20:
Actually, you're after `GIMBAL_DEVICE_ATTITUDE_STATUS` - sorry - misread that. So you should not need to request it - if you're connected that is supposed to be streamed by the gimbal all the time. @julianoes is there any special trigger you would expect for this? Does/should the gimbal device need to connect to some manager before it will emit the message (my assumption is not, but that the implementation might be wrong)?

Either way though, you will need to use the messages in mavlink/mavlink in order to be able to inspect the  GIMBAL_DEVICE_ATTITUDE_STATUS

FabianEP11, 2025-03-21:
Hi,

Yes, the issue is specifically with receiving GIMBAL_DEVICE_ATTITUDE_STATUS.
I am using a Pixhawk 6X with a real drone, and I'm running MAVLink Version 2.

For testing purposes, when I use a laptop and run QGroundControl, since it forwards the MAVLink messages, I can successfully receive the gimbal values at all times using this connection:
`master = mavutil.mavlink_connection('udp:localhost:14445')`

However, I am not planning on using QGroundControl for the final setup. Instead, I am connecting directly to the Pixhawk controller from a Jetson Nano running a Python script using this connection:
master = mavutil.mavlink_connection("/dev/ttyACM0", baud=115200)
which eventually starts receiving values after I start / kill the script multiple times. 

I suspect that the issue might be related to not using QGroundControl. Perhaps the GIMBAL_DEVICE_ATTITUDE_STATUS is not streamed by default when there isn't a MAVLink manager (like QGroundControl) forwarding the messages. Would this be the case?

If so, do you think there's a way to configure the Pixhawk to ensure the gimbal attitude status is streamed continuously?

Thank you for your help!

julianoes, 2025-03-23:
Are you forwarding between the mavlink instances in PX4?

What's the output of `mavlink status`?

And what's the output if `gimbal status` in the mavlink shell? And do the test commands work, e.g. `gimbal test yaw 45`?

FabianEP11, 2025-03-24:
Hi, 
I'm not entirely sure what you mean by "forwarding between the MAVLink instances in PX4," but I believe I am forwarding, as the connection I'm using (`/dev/ttyACM0`) shows Forwarding: On.

Here's the output from the mavlink status command:

```
Waiting for drone to connect...
-- Connected to drone!
nsh> mavlink status

mavlink status

instance #0:
	mavlink chan: #0
	type:		GENERIC LINK OR 
 RADIO
	flow control: ON
	rates:
	  tx: 2077.3 B/s
	  txerr: 0.0 B/s
	 
  tx rate mult: 0.050
	  tx rate max: 1200 B/s
	  rx: 477.2 B/s
	  rx  loss: 0.0%
	Received Messages:
	  sysid:  1, compid:154, Total: 117653 
 (lost: 0)
	    msgid:    4, Rate:  1.7 Hz, last 1.90s ago
	    msgid:    0, Rate:  1.0 Hz, last 0.13s ago
	    msgid:  285, Rate: 10.2 Hz,  
last 0.04s ago
	  sysid:  1, compid:100, Total: 10409 (lost: 676)
	    
 msgid:    0, Rate:  0.9 Hz, last 0.38s ago
	FTP enabled: YES, TX enab 
led: YES
	mode: Normal
	Forwarding: On
	MAVLink version: 2
	transport  
protocol: serial (/dev/ttyS6 @115200)

instance #1:
	mavlink chan: #1
 
	type:		GENERIC LINK OR RADIO
	flow control: OFF
	rates:
	  tx: 0.0 B/s
	  txerr: 175.7 B/s
	  tx rate mult: 0.050
	  tx rate max: 100000 B/s
	  rx: 0.0 B/s
	  rx loss: 0.0%
	FTP enabled: YES, TX enabled: YES
	 
mode: Normal
	Forwarding: Off
	MAVLink version: 1
	transport protocol: UDP (14550, remote port: 14550)
	Broadcast enabled: YES

instance #2: 

	GCS heartbeat valid
	mavlink chan: #2
	type:		USB CDC
	flow control: 
 OFF
	rates:
	  tx: 22639.9 B/s
	  txerr: 0.0 B/s
	  tx rate mult: 1.0 
00
	  tx rate max: 100000 B/s
	  rx: 46.9 B/s
	  rx loss: 99.4%
	Recei 
ved Messages:
	  sysid:255, compid:230, Total: 426 (lost: 131)
	  sysi 
d:255, compid:  0, Total: 9 (lost: 2032)
	  sysid:245, compid:190, Tot 
al: 2546 (lost: 294024)
	    msgid:   76, Rate:  0.0 Hz, last 7.31s ago
	    msgid:  126, Rate:  0.1 Hz, last 0.06s ago
	    msgid:    4, Rate:  1.4 Hz, last 0.84s ago
	    msgid:    0, Rate:  5.7 Hz, last 0.39s ago
	FTP enabled: YES, TX enabled: YES
	mode: Onboard
	Forwarding: On
	MAVLink version: 2
	transport protocol: serial (/dev/ttyACM0 @2000000)
	ping statistics:
	  last: 0.20 ms
	  mean: 13.65 ms
	  max: 5649200.50 ms
	  min: 0.15 ms
	  dropped packets: 6742
nsh> 
```
Here's the output from the `gimbal status` command:
```
nsh> gimbal status

gimbal status
INFO  [gimbal] Input Selected
INFO  [gimbal] Input not s 
elected
INFO  [gimbal] Input: Test
  roll :   0 deg
  pitch:  10 deg  
 yaw  :   0 deg
INFO  [gimbal] Input: Mavlink (CMD_MOUNT)
INFO  [gimbal] Primary control:   1/1
INFO  [gimbal] Output: MAVLink gimbal protocol v2
  quaternion: [1.0 0.0 0.1 0.0]
  angular velocity: [nan nan nan]
  gimbal device compid found: 154
```
And here's the result from running the `gimbal test yaw 45` command:
```
nsh> gimbal test yaw 45
gimbal test yaw 45
```
However, the gimbal did not move after issuing this command

Thanks for your time and help! 
I really appreciate any other suggestions you might have to help resolve this

hamishwillee, 2025-03-26:
FYI only, 
- MAVLink packets are supposed to be forwarded across channels according to [routing rules](https://mavlink.io/en/guide/routing.html). So for example if you have a broadcast mavlink message (one with no target component/system id) coming in via a serial link, it should then be forwarded to other links such as WiFi. For targetted messages, you only forward them to interfaces you've already seen a heartbeat from for that component.
- ArduPilot does this, so things that are expected to get the message do.
- PX4 doesn't. Instead it allows you to configure that everything is forwarded on a particular channel using [MAV_X_FORWARD](https://docs.px4.io/main/en/peripherals/mavlink_peripherals.html#MAV_X_FORWARD) for that channel.

So basically the channel that is connected to your gimbal needs to set MAV_X_FORWARD (whatever X is the gimbal mavlink instance) and then any messages from that channel will be send to others, such as the one to which QGC is connected.

It does look like it is set for the MAV_1_FORWARD channel on which you have the gimbal connected, and PX4 is seeing messages from the gimbal - heartbeat, ping and `GIMBAL_DEVICE_ATTITUDE_STATUS` 

```
	  sysid:  1, compid:154, Total: 117653 
 (lost: 0)
	    msgid:    4, Rate:  1.7 Hz, last 1.90s ago
	    msgid:    0, Rate:  1.0 Hz, last 0.13s ago
	    msgid:  285, Rate: 10.2 Hz,  
```

@julianoes will doubtless be able to make more of your debug information. If it isn't moving for the gimbal test that is a bad sign.

julianoes, 2025-03-26:
Make sure to set the MAV_x_MODE to Gimbal (10) instead of Normal. I'm not sure that's enough to get it working though but please try that first.

hamishwillee, 2025-04-02:
@FabianEP11 Did you have any success from Julian's suggestions?

FabianEP11, 2025-04-02:
Hi, sorry I haven't been able to test those commands yet. Our team is making some modifications to the testbed, and I don't have the hardware with me at the moment. I'll get back to you once I've had a chance to test it