mavlink/mavlink #2243: 在上传的时候报错

Sunrisies, 2025-03-20:
[image]

hamishwillee, 2025-03-20:
I'm afraid I can only accept questions in English. This is a picture - you're going to have to explain what you're trying - what flight stack, are you trying your own code, and if so what does it look like, and so on.

closed 2025-03-20 by hamishwillee

Sunrisies, 2025-03-20:
```js
const writeMissionCount = async (count) => {
    const missionCount = new common.MissionCount();
    missionCount.targetSystem = 1;
    missionCount.targetComponent = 1;
    missionCount.count = 1; // 航点总数
    missionCount.missionType = 0; // 任务类型
    console.log('Mission Count:', missionCount);
    await sendMavlinkMessage(missionCount);
    await new Promise((resolve) => {
        const listener = (packet) => {
            const clazz = REGISTRY[packet.header.msgid];
            if (clazz) {
                const data = packet.protocol.data(packet.payload, clazz);
                if (data instanceof common.MissionAck) {
                    console.log('Mission Count:', data);
                    resolve()
                    mavlinkReader.off('data', listener);
                }
            }
        };
        mavlinkReader.on('data', listener);
    });
}
```
An error occurred during upload, using the node mavlink library. When starting to write the total number of routes, the result returned was a type equal to 15

hamishwillee, 2025-03-26:
Cheers.
1. What flight stack are you using? 
2. The ack return value is https://mavlink.io/en/messages/common.html#MAV_MISSION_OPERATION_CANCELLED - which as described in the [Mission protocol](https://mavlink.io/en/services/mission.html#cancel) indicates that the mission has been requested to cancel.

Sunrisies, 2025-03-28:
[image]
I wrote it according to node mavlink, but it keeps reporting 15. Reading the route is not a problem, and I also wrote it according to mavlink

hamishwillee, 2025-04-01:
What flight stack are you using?