mavlink/mavlink #2275: SET_ATTITUDE_TARGET (82)

KKJava1, 2025-05-20:
Does this type of command allow the drone to perform movement operations like moving forward, backward, up, down, left, or right? For example, if I input a "forward" command as {x:1, y:0, h:0}, how can I send this command to make the drone move forward?

hamishwillee, 2025-05-21:
No, attitude is "pose" or orientation in 3D space. 

There are a bunch of ways to move a drone, but you're more likely to use https://mavlink.io/en/messages/common.html#SET_POSITION_TARGET_LOCAL_NED for that.

The best way to move a drone depends on your drone flight controller, the vehicle type, and so on. Not all flight controllers handle the same messages, and not all vehicle types handle the same messages or the same messages in the same way. 

The best place to ask questions is therefore the discussion boards for the vehicle flight stack that you are using, such as PX4 or ArduPilot, specifying the version of the flight stack, the type of vehicle, and what you are trying to do - i.e. precisely follow a path, follow a target that might be moving, or just "fly to some global X, Y cooordinate".

closed 2025-05-21 by hamishwillee

KKJava1, 2025-05-23:
> 不，姿态是 3D 空间中的“姿势”或方向。
> 
> 移动无人机的方法有很多种，但您更有可能使用https://mavlink.io/en/messages/common.html#SET_POSITION_TARGET_LOCAL_NED来实现。
> 
> 移动无人机的最佳方式取决于您的无人机飞行控制器、飞行器类型等因素。并非所有飞行控制器都会处理相同的消息，并且并非所有飞行器类型都会处理相同的消息，或者以相同的方式处理相同的消息。
> 
> 因此，提问的最佳地点是您正在使用的飞行器飞行堆栈的讨论板，例如 PX4 或 ArduPilot，指定飞行堆栈的版本、飞行器的类型以及您要做的事情 - 即精确地跟随路径、跟随可能正在移动的目标，或者只是“飞到某个全局 X、Y 坐标”。

ok,I know. I always use this command(84),but it's invalid.I am currently working with Java to directly control a drone. I have already completed the operations for takeoff, landing, and waypoint flight. I understand that in order to use the SET_POSITION_TARGET_LOCAL_NED command, the drone must first be set to offboard mode. However, I found in the documentation the MAV_CMD_DO_SET_MODE (176) command. Which MAV_MODE in this command can switch to offboard mode? Or, in the MAV_MODE enumeration, under which mode can the drone use the SET_POSITION_TARGET_LOCAL_NED command? thanks