mavlink/mavlink #2325: mavlink_helper.h fix

hwang12345, 2025-08-01:
I suggest modifying the attribute m_mavlink_status in mavlink_helper.h to define it as a global variable instead of a static local variable. The reason is as follows: when initializing the status using the mavlink_get_channel_status function during signature adaptation, it fails. After changing it to a global variable, it works normally.

hamishwillee, 2025-08-06:
Can you please repost in https://github.com/ArduPilot/pymavlink , ideally creating a PR request indicating how you would like it to look.
The toolchain is managed in that repo - here we mostly manage the message definitions.

hamishwillee, 2026-08-25:
Closing. Posted in wrong repo. If it is important, please repost.

closed 2026-08-25 by hamishwillee