mavlink/mavlink #2332: MAVLink C UDP example doesn't work

hamishwillee, 2025-08-13:
The MAVLink C example https://mavlink.io/en/mavgen_c/example_c_udp.html  can't be run as described (at least on windows).

Firstly, it doesn't build with 

```
gcc -std=c99 -I ../../include/common -o mavlink_udp udp_example.c
```

Because the the `#include <mavlink/common/mavlink.h>` doesn't point to the place the includes were just pointed to.

I modified this include to `#include <common/mavlink.h>` and built using `gcc -std=c99 -I ../../include -o mavlink_udp udp_example.c -Wno-address-of-packed-member`.

Then had error about the time include which I fixed by replacing the line as shown:

```
//#include <time.h>
#include <sys/time.h>
```

That builds with a warning.

Last fix was to hide some warnings with the command:
```
gcc -std=c99 -I ../../include -o mavlink_udp udp_example.c -Wno-address-of-packed-member
```

But I can't get this to connect to QGC. When I run the app I get endless repeats of:
```
sendto error: Invalid argument
recvfrom error: Resource temporarily unavailable
recvfrom error: Resource temporarily unavailable
recvfrom error: Resource temporarily unavailable
recvfrom error: Resource temporarily unavailable
```

Anyway, I'm doing in WSL and tried to connect but can't. Error also reported here: https://groups.google.com/g/mavlink/c/BbXePta7Ai4/m/vdQItiGZAwAJ

@julianoes I can't fix this because I don't know what is an acceptable fix for the code - i.e "are my changes OK".

If they are then I could merge, but still need to work out how to get this to connect. Perhaps the problem is that it assumes same network and WSL puts me in a new network.

hamishwillee, 2025-08-13:
I am running this in WSL2, but if you look at the linked user they have the same problem and may be using something different https://groups.google.com/g/mavlink/c/BbXePta7Ai4/m/vdQItiGZAwAJ

The build stuff shouldn't matter - if you can make a cmake target that allows this to build we can look at the actual connection as a separate issue.

julianoes, 2025-08-13:
Ah, there is already a [CMakeLists.txt](https://github.com/mavlink/mavlink/blob/master/examples/c/CMakeLists.txt). We should use that. As it says in the [README.md](https://github.com/mavlink/mavlink/blob/master/examples/c/README.md):

Start in root:
```
cmake -Bbuild -S. -DCMAKE_INSTALL_PREFIX=.
cmake --build build --target install
cd examples/c
cmake -Bbuild -S. -DCMAKE_PREFIX_PATH=../../
cmake --build build
./build/udp_example
```

And then it listens on 14550 (the docs are wrong there too), but it does not support `--help`.

I fixed the `sendto` error but the `recvfrom` error is correct if you are not connected.

See: https://github.com/mavlink/mavlink/pull/2333

hamishwillee, 2025-08-14:
Actually, doesn't completely fix this, at least on windows.

Install step in mavlink/mavlink works, but I get this error in the example folder:

```
D:\github\delme\mavlink\examples\c>cmake -Bbuild -H. -DCMAKE_PREFIX_PATH=$(pwd)/../../install
-- Building for: Visual Studio 17 2022
-- Selecting Windows SDK version 10.0.22621.0 to target Windows 10.0.26100.
-- The C compiler identification is MSVC 19.42.34435.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.42.34433/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
CMake Error at CMakeLists.txt:7 (find_package):
  By not providing "FindMAVLink.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "MAVLink", but
  CMake did not find one.

  Could not find a package configuration file provided by "MAVLink" with any
  of the following names:

    MAVLinkConfig.cmake
    mavlink-config.cmake

  Add the installation prefix of "MAVLink" to CMAKE_PREFIX_PATH or set
  "MAVLink_DIR" to a directory containing one of the above files.  If
  "MAVLink" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
```

hamishwillee, 2025-08-14:
Actually ignore that. We explicitly state this works on a unix-like system.

julianoes, 2025-08-14:
> According to ChatGpt this is because Windows is more strict about finding packages. I can't assess its solution though.

Haha. BS. I have to guess the CMAKE_PREFIX_PATH isn't correct, or the MAVLinkConfig.cmake we make doesn't work on Windows.

hamishwillee, 2025-08-14:
Ah, so this is connecting to an Autopilot!!! Docs say to QGC.

PR #2333 (2025-08-14): C example fixup
Closes #2332.